jangala-dev/devicecode-lua #30: Bus Publish/Publish Multiple

wobww, 2025-05-02:
I think we can tighten up the bus api by just removing the `publish_multiple` method.

I found this api not very intuitive:
```lua
local Bus = require 'bus'

local new_msg = Bus.new_msg

-- Publish a single message
connection:publish(new_msg({"foo", "bar", "fizz"}, "Hello World!", { retained = true }))

-- Publish multiple messages from a nested payload
connection:publish_multiple({"foo"}, {
    bar = {
        fizz = "Hello World!",
        buzz = "Another Message"
    }
}, { retained = false })
```

When you compare these two methods, it's a bit confusing, what are the different arguments for? 
- In `publish` the subtopic is a part of the message object which is passed as the first and only argument. 
- But `publish_multiple` takes just the subtopic as it's first arg? But then also takes another object as it's second argument?
What is this second argument for? It's not a message?

Also as a user of the bus, I think it's just simpler to have one method of publishing onto the bus. I'd imagine that to publish multiple messages, I would just call `publish` multiple times. Don't give me too many options!

lePereT, 2025-05-02:
Hey Rob - wonder if you could repost this issue on the bus repo? https://github.com/jangala-dev/lua-bus

wobww, 2025-05-02:
I've reposted on lua-bus, closing this one

closed 2025-05-02 by wobww