jangala-dev/devicecode-lua #31: Fibers Queues vs Go Buffered Channels

wobww, 2025-05-02:
Queues in fibers are compared to go's buffered channels, so what do the following (roughly equivalent bits of code) do?

**Go**
```go
package main

import "fmt"

func main() {
	ch := make(chan int, 2)

	go func() {
		fmt.Println("goroutine 1 started")
		ch <- 1
		ch <- 2
		ch <- 3
		fmt.Println("goroutine 1 blocked")
		fmt.Println("goroutine 1", <-ch)
		fmt.Println("goroutine 1 finished")
	}()

	go func() {
		fmt.Println("goroutine 2 started")
		fmt.Println("goroutine 2", <-ch)
		fmt.Println("goroutine 2", <-ch)
		fmt.Println("goroutine 2 finished")
	}()
}
```
**Lua Fibers**
```lua
package.path = "../../?.lua;../?.lua;" .. package.path

local fiber = require 'fibers.fiber'
local queue = require 'fibers.queue'

local q = queue.new(2)

fiber.spawn(function()
    print("fiber 1 spawned")
    q:put(1)
    q:put(2)
    q:put(3)
    print("fiber 1 blocked")
    print("fiber 1", q:get())
    fiber.stop()
end)

fiber.spawn(function()
    print("fiber 2 spawned")
    print("fiber 2", q:get())
    print("fiber 2", q:get())
    print("fiber 2 finished")
    fiber.stop()
end)

fiber.main()
```
**Answer**
- **Go** prints nothing
- **Lua Fibers** prints:
```
fiber 1 spawned
fiber 2 spawned
fiber 2 1
fiber 2 2
fiber 2 finished
```

I guess you could argue that they are not equivalent, but my main point is that it's not clear what the behaviour of the fibers queue is meant to be and comparisons with go maybe aren't that helpful.

There's some really good documentation and examples in fibers, but I'm wondering whether we could add more to make it more clear how these different primitives work without needing comparisons to go which ultimately doesn't work quite the same.

Also, adding more tests would be really helpful to show what the behaviour should be.

lePereT, 2025-05-02:
> wobww
>  created an issue

Thanks Rob, but there’s a few mistakes in your code!

A standard Go program exits on completion of `main()`. In the situation you’ve laid out where main spawn goroutines and you want to just keep the program running forever, you’d normally add `select{}` as the last thing in your `main` function. Without this the program immediately exits without these goroutines ever commencing.

You can achieve exactly the same (bad) behaviour in fibers:

```
package.path = "../../?.lua;../?.lua;" .. package.path

local fiber = require 'fibers.fiber'
local queue = require ‘fibers.queue'

local function main
     local q = queue.new(2)

     fiber.spawn(function()
        print("fiber 1 spawned")
        q:put(1)
        q:put(2)
        q:put(3)
        print("fiber 1 blocked")
        print("fiber 1", q:get())
     end)

    fiber.spawn(function()
        print("fiber 2 spawned")
        print("fiber 2", q:get())
        print("fiber 2", q:get())
        print("fiber 2 finished")
     end)

    fiber.stop()
end

fiber.main()

```

note that `fiber.stop()` doesn’t need to be called generally ever, def not at the end of every fiber. It’s akin to stopping the scheduler. Go implicitly calls it at the end of its main function.

lePereT, 2025-05-02:
> why doesn't fiber 1 unblock and finish?

because you're stopping the fibers engine with fiber.stop() :)

lePereT, 2025-05-02:
In terms of their theoretical foundations, concurrentML (the core of fibers) can achieve exact parity with golang's CSP (since CSP is a strict subset of CML's concurrency model).

For example, it was a trivial 1 to 1 task to translate `lua-bus` into `go-bus`.

As such I think it's helpful to use Golang's model as a nice stepping stone into fibers, but would welcome further thoughts!

lePereT, 2025-05-02:
Also here's versions that take advantage of our knowledge of the program to avoid the use of waitgroups completely:

```go
package main

import "fmt"

func main() {
	fmt.Println("main goroutine started")
	ch := make(chan int, 2)

	go func() {
		fmt.Println("goroutine 2 started")
		fmt.Println("goroutine 2", <-ch)
		fmt.Println("goroutine 2", <-ch)
		fmt.Println("goroutine 2 finished")
	}()

	ch <- 1
	ch <- 2
	ch <- 3
	fmt.Println("main goroutine blocked")
	fmt.Println("main goroutine", <-ch)
	fmt.Println("main goroutine finished")
}
```

```lua
package.path = "../../?.lua;../?.lua;" .. package.path

local fiber = require 'fibers.fiber'
local queue = require 'fibers.queue'

local function main()
    print("main fiber spawned")

    local q = queue.new(2)

    fiber.spawn(function()
        print("fiber 2 spawned")
        print("fiber 2", q:get())
        print("fiber 2", q:get())
        print("fiber 2 finished")
    end)

    q:put(1)
    q:put(2)
    q:put(3)
    print("main fiber blocked")
    print("main fiber", q:get())
    print("main fiber finished")

    fiber.stop()
end

fiber.spawn(main)
fiber.main()
```

wobww, 2025-05-02:
Hey Rich, thanks for your reply!

I wasn't necessarily trying to get anything to "work", I was just playing around with the examples comparing fibers and go and found that while conceptually they are similar, you are having to learn different rules about how to get fibers to do what you want and expect.

I'm not saying that we get rid of all references to go or anything, I just think we'd benefit from having more examples and more tests to make it clear what the expected behaviour of the fiber primitives are.

For example if you look at `test_queue.lua`, there's only one test in there, it uses an unbounded queue and it's actually quite hard to follow what's going on with the fibers under the hood. It would be great if we had some more tests that showcase bounded queues and queues being handled by multiple fibers.

Also, I removed the stop()s and I was wondering why does this hang?
```lua
package.path = "../../?.lua;../?.lua;" .. package.path

local fiber = require 'fibers.fiber'
local queue = require 'fibers.queue'

local q = queue.new(2)

fiber.spawn(function()
    print("fiber 1 spawned")
    q:put(1)
    q:put(2)
    q:put(3)
    print("fiber 1 blocked")
    print("fiber 1", q:get())
end)

fiber.spawn(function()
    print("fiber 2 spawned")
    print("fiber 2", q:get())
    print("fiber 2", q:get())
    print("fiber 2 finished")
end)

fiber.main()
```

lePereT, 2025-05-02:
yup - agree that there's lots of docs to write to allow users to get the most out of fibers. but conceptually its exactly the same as/a superset of go, and the primitives we've built should behave functionally equivalently to go counterparts :) in fact if you look at the `/examples` in lua-fibers the early examples are all translated directly from the Golang Tour

as to your question, take a look at the example in https://github.com/jangala-dev/devicecode-lua/issues/31#issuecomment-2847400441 for expected usage of `fiber.stop()`

`fiber.main()` is a blocking call to loop through the scheduler. as the scheduler can receive tasks from `epoll` based and other sources too. calling `fiber.stop()` allows `fiber.main()` to return and thus allows the script to exit. it's worth remembering that under the hood Go is doing pretty much exactly this, just implicitly at the end of its `main` function.

wobww, 2025-05-02:
Ok here's another example that I found [here](https://arc.net/l/quote/vxjigyal).

**Go**
```go
package main

import (
	"fmt"
	"time"
)

func write(ch chan int) {
	for i := 0; i < 4; i++ {
		ch <- i
		fmt.Println("successfully wrote", i, "to ch")
	}
	close(ch)
}
func main() {
	// creates capacity of 2
	ch := make(chan int, 2)
	go write(ch)
	time.Sleep(2 * time.Second)
	for v := range ch {
		fmt.Println("read value", v, "from ch")
		time.Sleep(2 * time.Second)

	}
}
```

**Fibers**
```lua
local function bounded_queue()
    local q = queue.new(2)

    local function write(kew)
        for i = 1,4 do
            kew:put(i)
            print("successfully wrote", i, "to queue")
        end
    end

    fiber.spawn(write(q))
    sleep.sleep(2)

    for i = 1,4 do
        local v = q:get()
        print("received", v, "from queue")
        sleep.sleep(2)
    end

    fiber.stop()
end

fiber.spawn(bounded_queue)
fiber.main()
```

While the go version writes
```
successfully wrote 0 to ch
successfully wrote 1 to ch
read value 0 from ch
successfully wrote 2 to ch
read value 1 from ch
successfully wrote 3 to ch
read value 2 from ch
read value 3 from ch
```

Fibers writes the following and then hangs
```
successfully wrote      1       to queue
successfully wrote      2       to queue
```

I tried a few different permutations on this and I couldn't get it to simulate what the go program does.

lePereT, 2025-05-02:
issue in line `fiber.spawn(write(q))` of your example. if you change to `fiber.spawn(function() write(q) end)` it'll work. remember you have to pass a _function_ to `fiber.spawn()` for it to in the underlying coroutine. instead you're calling `write(q)` directly and trying to pass its result

go's compiler adds that bit of syntactic magic which is arguably quite confusing :)

lePereT, 2025-05-02:
you can almost always just directly copy go's structure:

```lua
package.path = "../../?.lua;../?.lua;" .. package.path

local fiber = require 'fibers.fiber'
local queue = require 'fibers.queue'
local sleep = require 'fibers.sleep'

local function write(q)
    for i = 0,3 do
        q:put(i)
        print("successfully wrote", i, "to queue")
    end
end

local function main()
    local q = queue.new(2)
    fiber.spawn(function()
        write(q)
    end)
    sleep.sleep(2)

    for _ = 0,3 do
        print("read value", q:get(), "from ch")
        sleep.sleep(2)
    end
end

fiber.spawn(function() main() fiber.stop() end)
fiber.main()
```

there is a difference here with Go that we could choose to emulate or not. Go provide methods to close channels, whereas CML does not. Go overloads the closing of a channel as a form of in-band signalling: if a sender closes a channel then all subsequent reads will return nil-type values. on one hand, this is concise and useful, but also dangerous, as a send on a closed channel will panic. this limits some of the fan-in potential. it would be easy to make this change to mirror Go but...should we?

wobww, 2025-05-06:
Ah ok thanks for that Rich, I think I'm just unfamiliar with fibers so the examples really help.

I think what we have feels nice and simple, closing channels in Go always felt a bit risky and I'd rather we didn't introduce that here.

I'll close this issue, we can revisit adding more examples/documentation in another issue.

closed 2025-05-06 by wobww

lePereT, 2025-05-07:
> Ah ok thanks for that Rich, I think I'm just unfamiliar with fibers so the examples really help.
>
> I think what we have feels nice and simple, closing channels in Go always felt a bit risky and I'd [...]

Actually Rob, here's an idea - I've played around with some code on the main fibers repo (in a branch) that would allow us to do `nil` sends on channels. This would make the following example possible (integrating nicely with Lua's standard interators):

```lua
package.path = "../../?.lua;../?.lua;" .. package.path

local fiber = require 'fibers.fiber'
local queue = require 'fibers.queue'
local sleep = require 'fibers.sleep'

local function write(q)
    for i = 0,3 do
        q:put(i)
        print("successfully wrote", i, "to queue")
    end
    q:put(nil)
end

local function main()
    local q = queue.new(2)

    fiber.spawn(function()
        write(q)
    end)
    sleep.sleep(2)

    for element in function() return q:get() end do
        print(element)
    end
end

fiber.spawn(function() main() fiber.stop() end)
fiber.main()
```

It's quite nice. @wobww and @rslater-cs would love your thoughts on this. Channels in both CML and Go allow nil sends and ours don't.

wobww, 2025-05-08:
On one hand, I quite like this. I like the integration with Lua's iterators, and it also reminds me of the way you can do `for x := range chan` in go.

Also, it looks like there's an assertion currently when you try and put nil on a queue which maybe isn't ideal?
```lua
-- in queue.lua 
function ret:put_op(x)
  assert(x ~= nil)
  return ch_in:put_op(x)
end
```
I can see future bugs where nils are placed into a queue accidentally and errors being thrown, one way of handling this is that we allow nils in and out.

On the other hand...
What does it mean to receive nil from a channel? Is the channel "closed"? In go when the loop finishes it means the channel is closed so you shouldn't use it any more - but I'm guessing this wouldn't be the case for fibers. If I receive nil off of a queue, will it now always return nil? Or is there a chance that non-nil values will eventually be received?

Also, do we have any code that is relying on the fact that we aren't expecting a nil off the queue? Will this code then break if this changes, and will users of a queue now have to check for nils every time they receive off of a channel?

Maybe instead we could not let any nils on or off a channel? We could discard them on entry or return an error?

I guess there's lots of options and I'm leaning towards the "safer" ones. It would be a nice guarantee to know that you never receive a nil off a queue, would receiving nil ever be useful apart from signalling an end?

lePereT, 2025-05-08:
Sorry Rob - forgot to push my branch that enables nil sends: https://github.com/jangala-dev/lua-fibers/tree/nil_ch_put (this creates a nil safe FIFO queue for both channels and queues) All fibers tests pass with this change :)

"Also, do we have any code that is relying on the fact that we aren't expecting a nil off the queue?" Very good Q, I don't think we do, because `nil`s have never been in the space of possibilities (illegal for a sender to send).

I guess I quite like this halfway approach because it provides freedom for senders and receivers to explicitly negotiate the meaning of nils. eg sending temp sensor values - nil as no reading possible vs nil as stream processing/work complete