jangala-dev/devicecode-lua #40: Segment Fault Crash

rslater-cs, 2025-05-20:
Code runs for about 20 minutes before seg faulting
```
root@GetBox-6QnGq:~/build# luajit main.lua bigbox_v0-9
[TRACE 10:29:44] ./services/config.lua:35: Starting Config Service
[TRACE 10:29:44] ./services/hal.lua:266: Starting HAL Service
[TRACE 10:29:44] ./services/gsm.lua:694: Starting GSM Service
[TRACE 10:29:44] ./services/net.lua:598: Starting NET Service
[INFO  10:29:44] ./services/metrics.lua:337: Metrics Service Starting
[TRACE 10:29:44] ./services/system.lua:252: Starting System Service
main fiber sleeping
[TRACE 10:29:44] ./services/config.lua:18: Config: publishing config for:  net
[TRACE 10:29:44] ./services/config.lua:18: Config: publishing config for:  system
[TRACE 10:29:44] ./services/config.lua:18: Config: publishing config for:  metrics
[TRACE 10:29:44] ./services/config.lua:18: Config: publishing config for:  gsm
[TRACE 10:29:44] ./services/config.lua:30: Config: publishing config for: mainflux
[TRACE 10:29:44] ./services/hal/managers/modemcard.lua:31: Modem Detector: starting...
[TRACE 10:29:44] ./services/hal/managers/modemcard.lua:87: Modem Manager: starting
[TRACE 10:29:44] ./services/net.lua:85: NET: Config receiver starting
[TRACE 10:29:44] ./services/net.lua:103: NET: Interface listener starting
[TRACE 10:29:44] ./services/net.lua:411: NET: UCI manager starting
[TRACE 10:29:44] ./services/net.lua:525: NET: WAN monitor starting
[TRACE 10:29:44] ./services/net.lua:577: NET: Speedtest worker starting
[TRACE 10:29:44] ./services/net.lua:579: NET: Speedtest worker waiting for jobs
[TRACE 10:29:44] ./services/net.lua:371: NET: Config applier starting
[INFO  10:29:44] ./services/metrics.lua:199: Metrics Config Received
[TRACE 10:29:44] ./services/gsm.lua:620: GSM received config
[INFO  10:29:44] ./services/net.lua:93: NET: Config Received
[INFO  10:29:44] ./services/system.lua:39: System - nil: Disabling USB3
[INFO  10:29:44] ./services/net.lua:126: NET: Applying base firewall config
[INFO  10:29:44] ./services/net.lua:157: NET: Base firewall configured
[INFO  10:29:44] ./services/net.lua:161: NET: Applying base network config
[INFO  10:29:44] ./services/net.lua:188: NET: Base network configured
[INFO  10:29:44] ./services/net.lua:192: NET: Applying base mwan3 config
[INFO  10:29:44] ./services/net.lua:214: NET: Base mwan3 configured
[INFO  10:29:44] ./services/net.lua:221: NET: Applying network config for: adm
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added adm to dhcp ads interface
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added adm to firewall lan network
[INFO  10:29:44] ./services/net.lua:295: NET: Network config applied successfully for: adm
[INFO  10:29:44] ./services/net.lua:221: NET: Applying network config for: jan
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added jan to dhcp ads_adult interface
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added jan to firewall lan_rst network
[INFO  10:29:44] ./services/net.lua:295: NET: Network config applied successfully for: jan
[INFO  10:29:44] ./services/net.lua:221: NET: Applying network config for: int
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added int to dhcp standard interface
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added int to firewall lan network
[INFO  10:29:44] ./services/net.lua:295: NET: Network config applied successfully for: int
[DEBUG 10:29:44] ./services/net.lua:445: NET: Deferring config for mdm0 — waiting for modem primary
[DEBUG 10:29:44] ./services/net.lua:445: NET: Deferring config for mdm1 — waiting for modem secondary
[INFO  10:29:44] ./services/net.lua:221: NET: Applying network config for: wan
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added wan to firewall wan network
[DEBUG 10:29:44] ./services/net.lua:51: NET: Added wan_member to mwan3 def_pol use_member
[INFO  10:29:44] ./services/net.lua:295: NET: Network config applied successfully for: wan
[DEBUG 10:29:44] ./services/net.lua:455: NET: Committing changes for: network
[DEBUG 10:29:44] ./services/net.lua:455: NET: Committing changes for: firewall
[DEBUG 10:29:44] ./services/net.lua:455: NET: Committing changes for: dhcp
[DEBUG 10:29:44] ./services/net.lua:455: NET: Committing changes for: mwan3
That took:      0.015609929999982
[TRACE 10:29:44] ./services/hal/managers/modemcard.lua:59: Modem Detector: detected at: /org/freedesktop/ModemManager1/Modem/5
[TRACE 10:29:44] ./services/hal/managers/modemcard.lua:59: Modem Detector: detected at: /org/freedesktop/ModemManager1/Modem/6
[TRACE 10:29:44] ./services/hal/drivers/modem.lua:412: hal - Modem State Monitor - 869710032130530: Started for 869710032130530
[TRACE 10:29:44] ./services/hal/drivers/modem.lua:496: Modem - 869710032130530: Command Manager started
[INFO  10:29:44] ./services/hal/drivers/modem.lua:223: quectel eg25 g detected at: /org/freedesktop/ModemManager1/Modem/6
[TRACE 10:29:45] ./services/hal/drivers/modem.lua:412: hal - Modem State Monitor - 867929068986928: Started for 867929068986928
[TRACE 10:29:45] ./services/hal/drivers/modem.lua:496: Modem - 867929068986928: Command Manager started
[DEBUG 10:29:45] ./services/net.lua:468: NET: WAN state change wan online
[INFO  10:29:45] ./services/net.lua:474: NET: network wan : newly online, scheduling speedtest
[INFO  10:29:45] ./services/gsm.lua:558: Modem Capability 2 detected
[TRACE 10:29:45] ./services/gsm.lua:233: GSM - Autoconnect (secondary): Started
[TRACE 10:29:45] ./services/gsm.lua:257: GSM - Autoconnect (secondary): State recieved: connected
[INFO  10:29:45] ./services/net.lua:490: NET: Interface wwan1 resolved for modem secondary
[... 450 lines cut ...]
    rssi: 0
main fiber sleeping
main fiber sleeping
modem:
  1:
    rsrq: 5
main fiber sleeping
main fiber sleeping
modem:
  1:
    rsrq: -5
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
main fiber sleeping
Segmentation fault
```

cpunt, 2025-05-28:
Might be a coincidence but mine seg faulted while inserting a sim card
```
[DEBUG 17:12:17] ./services/hal/drivers/modem.lua:115: [DEBUG Wed May 28 17:12:17 2025] ./services/hal/drivers/modem.lua:115: 1
Publishing stats message	gsm.modem.secondary.sim
Publishing stats message	gsm.modem.secondary.signal.rssi
Publishing stats message	gsm.modem.secondary.operator
Publishing stats message	gsm.modem.secondary.iccid
main fiber sleeping, zombies:	2	open fds:	54	processes:	170
Publishing stats message	gsm.modem.primary.iccid
Publishing stats message	gsm.modem.primary.sim
Publishing stats message	gsm.modem.primary.signal.snr
Publishing stats message	gsm.modem.primary.signal.rssi
Publishing stats message	gsm.modem.primary.signal.rsrp
Publishing stats message	gsm.modem.primary.signal.rsrq
Publishing stats message	gsm.modem.primary.operator
main fiber sleeping, zombies:	2	open fds:	54	processes:	170
Publishing stats message	gsm.modem.primary.iccid
Segmentation fault
```

lePereT, 2025-06-03:
Marking as fixed due to https://github.com/jangala-dev/devicecode-lua/commit/b828b6113693b7a9283c0774fe01b1e48e2c7500 which brings in the following change in lua-fibers https://github.com/jangala-dev/lua-fibers/commit/3c5e86a00c6ab6d26cbfcdbb245e34e79695cf1f

closed 2025-06-03 by lePereT