jangala-dev/devicecode-lua #50: Modem 2 fails to register

cpunt, 2025-05-28:
Modem 2 getting stuck in enabled state after taking sim out:
```
[WARN Wed May 28 17:14:53 2025] ./services/gsm.lua:185: GSM - Autoconnect (secondary): Modem failed to register
```

```
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: enabled
           |             power state: on
           |             access tech: umts
           |          signal quality: 31% (cached)
```

cpunt, 2025-06-16:
I've experienced this behaviour again: 
[image]

Full output:
```
root@OpenWrt:~# mmcli -m /org/freedesktop/ModemManager1/Modem/0
  -----------------------------------
  General  |                    path: /org/freedesktop/ModemManager1/Modem/0
           |               device id: 9b6fb32f01777cb1853d78e5f993c51b38e63f4b
  -----------------------------------
  Hardware |            manufacturer: QUALCOMM INCORPORATED
           |                   model: QUECTEL Mobile Broadband Module
           |       firmware revision: EG25GGBR07A07M2G
           |          carrier config: ROW_Generic_3GPP
           | carrier config revision: 0501081F
           |            h/w revision: 10000
           |               supported: gsm-umts, lte
           |                 current: gsm-umts, lte
           |            equipment id: 860195054766628
  -----------------------------------
  System   |                  device: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4
           |                 drivers: qmi_wwan, option1
           |                  plugin: quectel
           |            primary port: cdc-wdm1
           |                   ports: cdc-wdm1 (qmi), ttyUSB4 (qcdm), ttyUSB5 (gps), 
           |                          ttyUSB6 (at), ttyUSB7 (at), wwan1 (net)
  -----------------------------------
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: enabled
           |             power state: on
           |             access tech: gsm
           |          signal quality: 70% (cached)
  -----------------------------------
  Modes    |               supported: allowed: 2g; preferred: none
           |                          allowed: 3g; preferred: none
           |                          allowed: 4g; preferred: none
           |                          allowed: 2g, 3g; preferred: 3g
           |                          allowed: 2g, 3g; preferred: 2g
           |                          allowed: 2g, 4g; preferred: 4g
           |                          allowed: 2g, 4g; preferred: 2g
           |                          allowed: 3g, 4g; preferred: 4g
           |                          allowed: 3g, 4g; preferred: 3g
           |                          allowed: 2g, 3g, 4g; preferred: 4g
           |                          allowed: 2g, 3g, 4g; preferred: 3g
           |                          allowed: 2g, 3g, 4g; preferred: 2g
           |                 current: allowed: 2g, 3g, 4g; preferred: 4g
  -----------------------------------
  Bands    |               supported: egsm, dcs, pcs, g850, utran-1, utran-4, utran-6, utran-5, 
           |                          utran-8, utran-2, eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, 
           |                          eutran-7, eutran-8, eutran-12, eutran-13, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-38, eutran-39, 
           |                          eutran-40, eutran-41, utran-19
           |                 current: egsm, dcs, pcs, g850, utran-1, utran-4, utran-6, utran-5, 
           |                          utran-8, utran-2, eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, 
           |                          eutran-7, eutran-8, eutran-12, eutran-13, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-38, eutran-39, 
           |                          eutran-40, eutran-41, utran-19
  -----------------------------------
  IP       |               supported: ipv4, ipv6, ipv4v6
  -----------------------------------
  3GPP     |                    imei: 860195054766628
           |           enabled locks: fixed-dialing
           |             operator id: 23410
           |            registration: idle
  -----------------------------------
  3GPP EPS |    ue mode of operation: csps-2
           |     initial bearer path: /org/freedesktop/ModemManager1/Bearer/0
           |  initial bearer ip type: ipv4v6
  -----------------------------------
  SIM      |        primary sim path: /org/freedesktop/ModemManager1/SIM/0
  -----------------------------------
  Bearer   |                   paths: /org/freedesktop/ModemManager1/Bearer/1
```

rslater-cs, 2025-06-30:
Related to #75, will come back to this if problem persists after FW upgrade

cpunt, 2025-07-03:
Temp fix:
https://github.com/jangala-dev/devicecode-lua/pull/92/files

closed 2025-07-16 by cpunt