jangala-dev/devicecode-lua #51: Net not reporting when modem back online?

cpunt, 2025-05-28:
Behaviour - removed sim card and reinserted. When back online net doesn't register:
[image]

```
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: connected
           |             power state: on
           |             access tech: umts
           |          signal quality: 44% (cached)
```

cpunt, 2025-06-17:
It seemed to of paused the interface:
```
root@OpenWrt:~# mwan3 interfaces
Interface status:
 interface wan is online 00h:01m:18s, uptime 00h:14m:37s and tracking is active
 interface mdm1 is offline 00h:00m:00s, uptime 00h:35m:06s and tracking is active
 interface mdm0 is offline 00h:00m:00s, uptime 00h:29m:51s and tracking is active

root@OpenWrt:~# ifup mdm1
root@OpenWrt:~# mwan3 interfaces
Interface status:
 interface wan is online 00h:06m:13s, uptime 00h:19m:32s and tracking is active
 interface mdm1 is offline and tracking is paused
 interface mdm0 is offline 00h:00m:00s, uptime 00h:34m:46s and tracking is active
```

rslater-cs, 2025-07-10:
Sim was out of data, closing