jangala-dev/devicecode-lua #52: Cqueues issues

cpunt, 2025-05-29:
https://github.com/wahern/cqueues/issues/210

```
[WARN  17:00:27] ./services/ui.lua:145: [WARN  Wed May 28 17:00:27 2025] ./services/ui.lua:145: SSE: Client disconnected or write error: /usr/lib/lua/cqueues/socket.lua:579: exceeded unchecked error limit (Broken pipe)
[INFO  17:00:27] ./services/ui.lua:155: [INFO  Wed May 28 17:00:27 2025] ./services/ui.lua:155: SSE: Client disconnected stats
Publishing stats message	gsm.modem.primary.operator
[WARN  17:00:27] ./services/ui.lua:339: [WARN  Wed May 28 17:00:27 2025] ./services/ui.lua:339: onstream on http.h1_stream{connection=http.h1_connection{type="server";version=1.1};state="closed"} failed: /usr/lib/lua/cqueues/socket.lua:579: exceeded unchecked error limit (Broken pipe)
```

cpunt, 2025-06-23:
It looks like we're not disconnected past clients properly. When we refresh the page we still send messages to the "old" client

m-imranali, 2025-06-24:
That is correct, the reason for it not being able to disconnect the past clients and stop sending messages before getting error is because the "stream" does not update the connection state from "half closed (remote)" to closed until later on after receiving enough errors with `stream:write_chunk`. This behavior enough is not responsible for the UI to freeze, the cause of UI freeze is the Event Loop Leakage where the `bus_listener` gets blocked and is not publishing messages anymore for SSE work on even though streams are alive and the client is present.