jangala-dev/devicecode-lua #71: Consider avoiding full mwan3 restarts when applying config changes to reduce interface disruption

cpunt, 2025-06-23:
When updating the /etc/config/mwan3 file, we're currently restarting the mwan3 service. While this ensures a clean reload, it also causes all tracked interfaces to briefly flap, leading to a burst of hotplug.mwan3 events.

This results in disconnects and reconnects for interfaces that were otherwise stable, which can introduce unnecessary backhaul churn, especially noticeable when a new SIM is inserted while other interfaces are online.

Example event output during a restart:
```
{ "hotplug.mwan3": {"action":"connected","interface":"wan","device":"eth0.4"} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"wan","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm1","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"wan","device":"eth0.4"} }
{ "hotplug.mwan3": {"action":"connecting","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"wan","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm1","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"wan","device":"eth0.4"} }
{ "hotplug.mwan3": {"action":"connecting","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"wan","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm1","device":""} }
{ "hotplug.mwan3": {"action":"ifdown","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"connecting","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"connected","interface":"wan","device":"eth0.4"} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"ifup","interface":"mdm0","device":"wwan0"} }
{ "hotplug.mwan3": {"action":"connecting","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"wan","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm1","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm0","device":"wwan0"} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"connected","interface":"wan","device":"eth0.4"} }
{ "hotplug.mwan3": {"action":"connecting","interface":"mdm0","device":"wwan0"} }
```

We might want to explore using mwan3 reload instead of a full restart after config changes (i.e. after uci commit mwan3). This should allow us to apply changes without bringing down interfaces that haven't changed, leading to smoother transitions and fewer interruptions.

rslater-cs, 2025-07-23:
To further reduce turning interfaces on and off we should think about caching speed test results by modem and sim identifiers, if a modem happens to drop off periodically and instantly reconnects we don't need to redo a speedtest since it is the same modem+sim pairing

cpunt, 2025-07-29:
Every time an sim is taken out it disconnects the other interfaces causing a speedtest to be run:
```
{ "hotplug.mwan3": {"action":"disconnecting","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm1","device":"wwan1"} }
{ "hotplug.mwan3": {"action":"ifdown","interface":"mdm1","device":""} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm0","device":"wwan0"} }
{ "hotplug.mwan3": {"action":"disconnected","interface":"mdm0","device":""} }
{ "hotplug.mwan3": {"action":"connected","interface":"mdm0","device":"wwan0"} }
```

Seems to be getting stuck in a loop where the following will happen.

Interface will come online

cpunt, 2025-07-29:
Seems to be getting stuck in a loop where the following will happen:

* Interface will come online
* Speed test will be run
* Interface will be restarted because speed test run